alpriest/EnergyStats-Android #2: Stats month does not display correctly

TonyM1958, 2023-05-31:
Select Stats and Month. Browse months, the order is January, March, March, May, May. February and April sometimes appear in the drop down but are not displayed when selected.

This is February:

[image]

[image]

This is April:

[image]

[image]

[image]

[image]

alpriest, 2023-05-31:
Thanks, looks like the app was subtracting 1 month from the current date (31st May), getting 31st April which is invalid and sticking with May. Not sure why it didn't crash completely.

Fix just pushed to the Play store.

closed 2023-05-31 by alpriest

TonyM1958, 2023-05-31:
Thanks - obvious now you said it, that these are the months that don't have 31 days! Just luck I suppose that I looked at it today...