alpriest/EnergyStats-Android #3: Load power with second inverter

TonyM1958, 2023-05-31:
When an other solar inverter is connected to CT2, the load power sent by the inverter, detected by CT1 is incorrect as the second inverter is adding power into the load / grid.

To correct this, you need to add the power reported by CT2 to the load power reported by the inverter to give the correct load power.

You may also want to add the power from CT2 as another solar source and possibly integrate and add this into the total yield to get the overall yield from both inverters.

alpriest, 2023-06-11:
I've been looking at this but can't see the CT2 load power reported from FoxESS. Do you have any idea which parameter it is?

TonyM1958, 2023-06-11:
Looking at foxesscloud.com, it is the power from the second inverter in the power flow. In the fox app, it's the power from the top inverter.

In the drop down list of variables, there are 2 called MeterPower and Meter2Power. The first is also known as Grid_CT / CT1 and the 2nd is CT2 in the HA integrations for single phase inverters.

There is also R, S and T versions of LoadPower and MeterPower for 3 phase inverters. There is an option to connect a second meter to a 3 phase inverter (equivalent to CT1 and CT2) but I don't currently have a good example of how these might affect the 3 phase inverter load power.

TonyM1958, 2023-06-11:
I scanned the fox cloud raw data and pulled out the json variables list:

"variables": ["pv1Volt", "pv1Current", "pv1Power", "pvPower", "pv2Volt", "pv2Current", "pv2Power", "pv3Volt", "pv3Current", "pv3Power", "pv4Volt", "pv4Current", "pv4Power", "epsPower", "epsCurrentR", "epsVoltR", "epsPowerR", "epsCurrentS", "epsVoltS", "epsPowerS", "epsCurrentT", "epsVoltT", "epsPowerT", "RCurrent", "RVolt", "RFreq", "RPower", "SCurrent", "SVolt", "SFreq", "SPower", "TCurrent", "TVolt", "TFreq", "TPower", "ambientTemperation", "boostTemperation", "invTemperation", "chargeTemperature", "batTemperature", "dspTemperature", "loadsPower", "loadsPowerR", "loadsPowerS", "loadsPowerT", "generationPower", "feedinPower", "gridConsumptionPower", "invBatVolt", "invBatCurrent", "invBatPower", "batChargePower", "batDischargePower", "batVolt", "batCurrent", "meterPower", "meterPower2", "meterPowerR", "meterPowerS", "meterPowerT", "SoC", "ReactivePower", "PowerFactor"]

From this, it looks like you need the one called "meterPower2".

alpriest, 2023-06-12:
I just tried an account with multiple inverters (H1-6.0 and F3000). For that setup the H1-6.0 the `meterPower2` is showing negative values so if I sum that with the house load I should get the correct house from that inverter I think. I'll give that a go this week.

TonyM1958, 2023-06-12:
The example I've seen using HA data: load power from the inverter was negative and CT2 was positive, so adding CT2 to inverter load power produced the correct load power in HA.

Fox appear to do some fix ups somewhere between the inverter and the cloud data, as the user was saying the load power in the cloud was correct when raw data in HA was negative. But, if this is correct, why are the values in Energy Stats wrong...

Best to look at the data displayed in the Fox app and also foxesscloud alongside your query data to see if you can work out what to adjust to align things?

alpriest, 2023-08-29:
I've been trying this today based on our FB conversation. I have access to an account with two inverters (F3000 and H1-6.0-E).

I can get Energy Stats to mirror FoxCloud 100% for both inverters in the energy flow from the `loadsPower` variable without using the `meterPower2` variable. Energy Stats doesn't display the second inverter on the flow screen which is providing power.

If I sum the `meterPower2` (negative) and `loadsPower` (positive) then I get a loads total different than Fox cloud so I think i'm missing something.

TonyM1958, 2023-08-29:
OK, I just replied to your previous comment on FB, but let's continue here as its all a bit technical for a general audience.

Obviously, there is something about the setups with 2 inverters we don't understand, as the screen shots showed -ve load power but you are not seeing that. May need to put out a release that doesn't change loads Power and ask for feedback to try to narrow things down.

I wonder if it's as complex as whether the solar inverter is Fox and has a datalogger so they make adjustments to the cloud data for inverters in the same site? How many inverters do you see on your test account?

alpriest, 2023-08-29:
Here's a snapshot of the two inverters on my secondary account (a helpful user provided me their login to narrow down this stuff).

[image]

Currently i'm reading home usage as:

`raws.currentValue(for: "gridConsumptionPower") + raws.currentValue(for: "generationPower") - raws.currentValue(for: "feedinPower")`

This seems overly complex as `raws.currentValue(for: "loadsPower")` seems to get the same value as Fox. I think you're right to put out a new version with this and the change for `solar = hasPV ? pvPower : meterPower2` which will bring back the solar feed for multiple inverter users.

TonyM1958, 2023-08-29:
Generating is inverter output, so I think H1 is using excess generation to charge the battery in that snapshot.

One option would be to have a setting, off by default,  to add meterPower2 to loadsPower for people who's load goes negative. They can switch it on if needed, rather than trying to work out all the different scenarios?

TonyM1958, 2023-08-29:
BTW, in your Home usage calculation above, the missing element is meterPower2 or generationPower from the solar inverter(should be same thing) as these are all connected to the same node. I wonder if your value goes negative because this isn't included, where Fox are adding this to their loadsPower in the cloud because they see both inverters in the same site.

I think you may still see a negative value if the second inverter is not in the Fox cloud?

alpriest, 2023-08-29:
My second account has both inverters in the Fox cloud.

The H1-6.0-E shows
meter2Power: -1.228
pvPower: 1.6219999999999999
loadPower: 0.722

[image]

And this is what I'm showing on my dev build.

[image]

This obviously feels wrong that the solar is generating less than the battery amount. If I add the absolute `meterPower2` to the 'pvPower' for solar then I get this which seems closer.

[image]

I suspect as you say it might be best to have some of this as toggles for a user to select for their own configuration as there are clearly complexities not handled by the cloud for some of this.

TonyM1958, 2023-08-29:
One of the problems with meterPower2 is that there isn't a good way to tell if CT2 is on the right way around. I suspect Fox detects which way around the clamp is and reverses the polarity - so in the example above meterPower2 should be +1.228 not -1.228 ?

TonyM1958, 2023-08-29:
Again, most flexible option might be a switch to flip polarity of CT2 / meterPower2.  Could be a garage with an EV charger on the other end...

Not sure if you would need to add CT2 to load power and zero generation if it goes negative when the polarity is correct for generation as I don't have any examples of this.

TonyM1958, 2023-08-29:
So, I looked at this and decided there was no reliable way to detect if CT2 was reversed. For my Fox Cloud python code, I'm going to add a global setting you can configure that flips CT2 if required. 

Looking at the load, I decided, if you need to add CT2 to load, use the abs value for this, so it adds when generating and if the load reverses. For total PV, I ignore negative CT2 values and just add positive values to PV generation. This means, if there is a load at the far end of the cable CT2 monitors, this adds to the load when negative but not to the generation...

I dont think there is 100% right answer? Fix what works for you and put it out there - there are too many use cases and not enough test cases for everything to be fully tied down.

alpriest, 2023-08-31:
I’ve just pushed 1.58 to the Play Store review. It updates to use pvPower + meterPower2 (inverted based on a config toggle).

Al

TonyM1958, 2023-08-31:
Thanks - I've downloaded and installed this. The power values look correct for my setup, compared to what Home Assistant is showing me. I have H1-6.0-E, CT2 is not connected.

HA is telling me my solar yield for today is 3.1 kwh and EnergyStats is telling me 6.1 kwh though - so it look like it might be double counting the yield?

TonyM1958, 2023-08-31:
Also - when I go into inverter settings and toggle 'show inverter icon', 'invert CT2' toggles with it - either they are the same or the wrong toggle is being displayed?

TonyM1958, 2023-08-31:
if you are doing updates really minor, but the device ID text clashes with the label. Suggest either wrapping across 2 lines or not showing the device ID as I don't think the end user can really do anything with this?

alpriest, 2023-08-31:
The toggle was a copy/paste error. I duplicated the toggle code but forgot to update the underlying variable it modified. Fixed in 1.59.

Can you attach a screenshot of what you mean with the clashing please, it wraps two lines already according to the previews I see in Android Studio.

[image]

The Module Serial Number is returned in the response from https://www.foxesscloud.com/c/v0/device/addressbook?deviceID=<###>. I'm not sure what these things actually are but was asked by a couple of people to show them.

TonyM1958, 2023-08-31:
[image]

Here's what it looks like for me - just looks a bit untidy? Don't know if you could wrap in the middle or add a space at the end of label so it looks better?

The module serial number is present in the data logger info and I see it when I query my inverter device info but it shows as unknown here.

alpriest, 2023-09-04:
Can you try log out and back in please to see if that resolves the module SN issue?The latest version has a small gap between the label and device ID, which may help a little. I should also vertically align the rows.

TonyM1958, 2023-09-04:
No problem, that definitely helps. Presumably something was being cached?

[image]

I've seem you tell people to do this a few times, is there anyway to force a refresh?

alpriest, 2023-09-04:
There’s no easy way unfortunately. I added an additional stored field (hasBattery). On iOS if that doesn’t exist in the stored version of the Inverter object then it fails to fetch it, I detect that and then fetch the inverter again. On Android it silently sets the Boolean to false when it doesn’t find that field, so I can’t differentiate between a legitimate false and a bad decode becoming false.

TonyM1958, 2023-09-04:
Do you cache the user credentials in the back ground? I store the time when I get the token and refresh the token if its more than 1 hour old. That seems to avoid the problem Fox has when you get token no longer valid.

Just wonder if you could do something similar and force a logout to clear the cache and then login in the background periodically.  It doesn't seem to take long, so might not be a big overhead?

Obviously, this might fail if someone changed their password but you'd need to ask them to enter it again anyway.

alpriest, 2023-09-04:
Yep, the credentials are cached and then I relogin silently if I get a 401 Unauthorised from the server and retry the original request. So if Fox change their token validity period it will automatically refresh at the new duration.

Due to the way I stored the data if I log someone out then they would lose their battery capacity override as it's stored in the same model, and then it becomes impossible to differentiate between the user-overridden value and a potentially different value from the server. Easiest just to reply to FB for folk to relogin in this instance then they're in control.

alpriest, 2023-09-05:
> So, presumably, they loose the over ride anyway when they logout and back in again. Guess it's hard to tell how many people actually use the over ride - I did to start with but it always seem to get the right value these days.
>
> You can't grab the over ride, logout, login and restore it? Or have a button to remove cached data that logs out and back in again - just to avoid putting people that can't remember their password through the login again?

I’ve modified the apps to externalise the storage of the override value and reload the device list and firmware versions on cold start.

Anyone who had overridden it will lose their values when they upgrade but they can just enter it again.

TonyM1958, 2023-09-10:
To summarise what we found looking at Sam's data: if CT2 is the wrong way around, Fox calculates the loadsPower incorrectly and you have to reverse CT2 to fix this.

When loadsPower works as expected, meterPower2 in the cloud reads a negative value - so you need to invert the value to be able to show the solar generation correctly. as loadsPower doesn't work the wrong way around, you may as well remove the option to flip CT2 and hard code the inversion?

This is what I have now done in my code to generate PV Output. I don't invert the raw values that are returned by Fox as I think this would just add to any confusion about that the underlying data looks like. I only use the negative values if I calculate the energy generated via CT2 and ignore any postive values.

Does that make sense?

TonyM1958, 2023-09-10:
I've asked Richard Bagshaw for some data from his system as it has HA and 2 x inverters with CT2 connected. Here's what it looks like.

This is HA showing the raw CT2 sensor data and the Load Power after CT2 has been added to it. Here's CT2 is +ve when the second inverter is generating:

[image]

This is the same data collected from the Fox cloud in EnergyStats:

[image]

It's quite a dull day today, but you can clearly see that Fox has inverted Meter2Power compared to the data coming into HA directly from the inverter. Both have Load Power fixed compared to what is coming directly from the inverter.

So, this is why I was confused before... This confirms that the default behaviour for data from Fox Cloud should be to invert CT2 to get the solar generation from the second inverter.

Richard has both inverters available in Energy Stats so I wondered what the best way might be to display data in EnergyStats?

One option to consider might be, instead of the switch to flip CT2 for the main inverter, you could have a switch to enable / disable inclusion of CT2 in the solar generation you show in the power flow screen. By default, the switch would be on (so it shows generation associated with AC inverters), but users with 2 inverters would have the option to only show solar generation from the main inverter if they wanted?

Alternatively, you could show a break down of the solar generation - what power is coming from PV and what power is coming via CT2 (e.g. when inverted CT2 is making a +ve contribution to generation) on the power flow screen so it's more obvious what people are looking at?

alpriest, 2023-09-13:
I think having the transparency of a CT2 source on the power flow page is probably the most clear. With an option to aggregate (as current) or not.

alpriest, 2023-10-11:
I've added a separate CT2 indicator in the latest version. I've also fixed a bug whereby the CT2 values weren't being inverted if your inverter hasPV == false.

I can't seem to determine when to show the CT2 feed automatically so if it's enabled then it appears.

closed 2023-10-11 by alpriest