alpriest/EnergyStats-Android #5: Report Reisdual on Power Flow?

TonyM1958, 2023-06-15:
I tell a few people to dig through the menus into the debug info to check the Residual value of the battery.

I've been tracking this for a few weeks and it always agrees with the BMS kWh Remaining obtained over RS485 in Home Assistant. Personally, I use this more than SoC to track what's in the battery and also to check the battery capacity.

Would it be useful to have a switch in the settings to display Residual on the Power Flow screen in addition to SoC?
Maybe something like 100% (13.82kWh)?

alpriest, 2023-06-16:
Residual is already available by tapping on the % on the Power Flow Screen.

closed 2023-06-16 by alpriest

TonyM1958, 2023-06-17:
Cool. There must be a few things like this that lots of people don't know about, like me. I looked at Parameters a few times before I realised I could click the button and select the parameters. Could do with a guide or tips and tricks that is a bit more prominent? You could add a link or links in the app and point to a wiki page here that contains a user guide? (Like you did for my fox cloud info). Happy to contribute, if that helps...

alpriest, 2023-06-17:
Yep, you’re right. My plan is to add some tips exactly as you say. I’ve not had a look for frameworks that could support this - ideally a rules based one that would let me add milestones (number of app runs, screen opens) and trigger based on those. The next version of iOS has just this but I’ve not looked if there’s something for Compose that could help.More than to happy to accept PRs.