alpriest/EnergyStats-Android #7: Self sufficiency calculation isn't correct

alpriest, 2023-06-29:
Sometimes shows > 100%.

TonyM1958, 2023-06-29:
Hi,

I think there are a few problems with the home consumption calculation:

a) Charge energy from solar is taken before energy out of the inverter, so that has no relationship to house load.
b) Charge energy added from the grid is included in the grid consumption but you can't differentiate it from solar.
c) Output energy from the inverter includes discharge energy from the battery, so there is double counting here.

So, for a variety of reasons, trying to calculate home consumption from the reported energy figures isn't going to work in different circumstances.

The simplest way to get home consumption is to use the figure that Fox has in the cloud data. Add the variable 'loads' to your report query and it will tell you the home consumption recorded directly by the inverter. This is not an approximation, so you can add this directly to the stats listed?

I'll take a look at the HA Energy Card and see if I can work out how it calculates self sufficiency.

TonyM1958, 2023-06-29:
There are 2 things the Energy Dashboard shows:
a) Net Returned to the Grid
b) Self Consumed solar energy

The energy dashboard calculation isn't completely accurate unfortunately as it uses raw solar generation data and lumps system losses into it's calculation of home consumpution. Also, we don't really have total solar generation, so it's hard to work out self consumed solar. So, I'm not 100% convinced this the model to follow?

The grid consumption and feed in energy are measured by the CT clamp and are self contained as they both relate to the main grid connection. As long as you can get Home Consumption from the Fox Cloud, here's what I think you can work out...

It is easy (and useful) to work out Net Generation (= Feed In - Grid Consumption in kWh) and should be correct when reported cumulatively by day, month and year. This should also (roughly as we are looking at different measurement devices) correspond to the supply meter import, export and net deltas for day, month or year.

For self-sufficiency, I think you can use Net Generation and Home Consumption to provide a meaningful metric:
a) If Net Generation > 0, you generated more than you consumed so you are  more than 100% self sufficient.
b) If Net Generation is -ve, you are consuming grid power. If Net Generation + Home Consumption <= 0, your self sufficiency is 0% as you are consuming more than you use (which can happen due to losses charging / discharging your battery in winter).
c)  If Net Generation + Home Consumption > 0, then your self sufficiency is (Net Generation + Home Consumption) / Home Consumption * 100% (N/A if Home Consumption is zero, just in case)

You can merge (a) into (c) if you think it's useful to report self-sufficiency > 100%.

Does that make sense?

alpriest, 2023-06-29:
Thanks. So I've add the `report` API `loads` variable to get home consumption rather than calculating it, and then updated the calculation. 

Today my loads is 3600w, grid 100w and feed-in of 3400w. Net generation is therefore 3300w, and home consumption of 3600w so I get 100% self sufficient. I think it might need to be labelled 'net self sufficiency' to be more accurate. So that looks reasonable.

However yesterday (cloudy), loads was 6500w, grid 300w, feed-in of 0w which self sufficiency suggests 0% which isn't correct as we generated solar and supplied the rest via battery.

I think it needs to take into account the `output energy` or `discharge energy - charge energy` but appreciate `output energy` includes battery discharge.

alpriest, 2023-06-29:
I wonder if a simpler approximation would be to show the inverse of grid consumption as a percentage of home usage. On the above figures today would be self-sufficiency of 99.97%.

Yesterday would be 99.95%.

This doesn't account for any feed in though which could be offsets the sufficiency.

TonyM1958, 2023-06-29:
Self sufficiency is really a long term thing - best viewed over a week, month or year. The problem with daily self sufficiency is that you can start the day with an empty battery and end the day with a full one, so this biases the net generation. From the point of view of thermodynamics, it's a closed system with net energy flow in and out of the grid connection with a variable level of energy stored in the battery on a short term basis.

The adjustment I would make would be to say Net Generation = Feed In - Grid Consumption + Charge - Discharge. This will adjust the daily amounts for what is put into or taken from the battery but won't affect the long term as charge - discharge will net off to zero (or to battery losses - depending how Fox measure this).

You could make this adjustment to all Net Generation calculations (easier) or just to daily Net Generation - just depends on what you think the results look like overall.

In terms of the self sufficiency calculation - I think it's down to what you are comfortable with and happy to explain. I'm sure once you put the figures out there, there will be some discussion and a concensus is likely to emerge? For me, if I'm a net exporter, then I'm 100% self sufficient not 99%?

The energy values are in Wh or kWh btw :-)

TonyM1958, 2023-06-29:
Just one more thing to watch... in the Fox raw data, there seems to be problems with Load power when CT2 is being used. I don't know if this also applies to the Load energy - hopefully, Fox fixed the Load energy if not the Load power. They have also had problems (as reported on the FB group) over the last week where people have more than 1 inveter connected.

I don't think this means you should do anything different as there arent any great alternatives to getting Load energy. Just keep an eye out for anyone with multiple inverters reporting the numbers are wrong... (but I don't think that is anything new??)

closed 2023-06-30 by alpriest

TonyM1958, 2023-07-07:
For info: https://www.home-assistant.io/blog/2023/07/05/release-20237/#energy-dashboard-self-sufficiency

New indicator added to HA energy dashboard. May be worth having a look / comparing to the values shown in HA. Looks like it may be using 100% - grid consumption / home consumption?

alpriest, 2023-07-07:
Yep, it’s doing the same as my absolute self sufficiency but taking into account battery filled from grid. The numbers from Fox lag behind HA so it’s hard to do a direct comparison.