alpriest/EnergyStats-Android #9: Residual does not display correctly on Power Flow screen

TonyM1958, 2023-07-02:
Not sure what changed in the last update re: Residual kWh display on the power flow page.

The % and kWh used to match what I was seeing in Home Assistant. Now, the SoC % matches but the Residual does not with Android app version 1.27:

- Current display in HA is 96% and 13.36kWh
- Debug data in EnergyStats says SOC 96 and Residual 13360.0, so matches HA
- Power flow screen says 96% and 13.17kWh.

Shouldn't it also say 13.36kWh on the power flow screen?

alpriest, 2023-07-02:
The residual on the power flow screen has always been estimated - the residual read from the Battery settings was only used to estimate the full battery capacity.

Makes sense to read the battery settings though on 5min refresh interval and use the residual amount.

TonyM1958, 2023-07-02:
OK, thanks. I think you are compounding calculation errors by estimating capacity and using this to recalculate residual. As you say, just better to display the figure the BMS reports alongside SoC.

Specified and current capacity can be different and are the basis for State of Health. This is explained a bit more in a couple of sections here: https://github.com/TonyM1958/HA-FoxESS-Modbus/wiki/Battery-Management