alpriest/EnergyStats-Android #10: Change settings for battery / work mode

TonyM1958, 2023-07-09:
Hi Al,

Don't know if this is of interest, but I'll put it here anyway...

I've updated my python sample code with a set of calls that allow you to set the charge times, min soc and work modes if required using the Fox cloud API: https://github.com/TonyM1958/FoxESS-Cloud

Not going to claim any prizes - my coding is pretty basic synchronus REST API calls but the URLs, parameters and data structures are all there embedded in the code and tested via some examples of use in Jupyter Lab.

This might be something you already looked at and discounted for Energy Stats?

alpriest, 2023-07-09:
Hi, i've not yet looked at this. It's on my todo list but pretty stacked with other work at the moment so i've not had time to investigate. Also i'm still considering the best place to integrate these sort of things without making the UI too busy. Maybe the settings would work as i've been improving the coding behind that recently to allow for navigation within that tab.

Thanks though i'll check out your work when I get around to it 👍

alpriest, 2023-08-01:
This is available in the latest version.

closed 2023-08-01 by alpriest

TonyM1958, 2023-08-01:
Installed this today, works great. Do think a minor mod to the status comment might help some users understand what it does and differentiate between force charge from grid, force charge (when times are not zero) and disabled (when times are zero)?

I think it's also worth a post on the FB group to highlight the new feature and get people to install the update? Great that people can do this all in one place without needing to go back to the Fox app.

alpriest, 2023-08-01:
Do you have a suggestion? The text by the Save button is intended to describe the selected behaviour.

I wasn't aware that anything happened when there are times set but the checkbox isn't set? This option was removed from the Fox app and now they always send "true" for it.

TonyM1958, 2023-08-01:
You could add a disable button that clears the times for each period (which is what I think Fox originally meant to do) but it may just be simplest to mirror the Fox app UI so users don't have to learn a new way but provide a better explanation of what the setting they have chosen actually does.