alpriest/EnergyStats-Android #14: Self-sufficiency estimates descriptions are incorrect

spco, 2023-08-25:
Hi @alpriest - thanks for this app, it's brilliant in so many ways and your work is really appreciated.

On v1.54, the selection of Net and Absolute Self-sufficiency estimates seems to be incorrectly labelled or described - I think they need swapping over.

When I select "Net", the description is `(export - import + batCharge - batDischarge)/ houseUsage`. However, the estimates shown are being calculated as `1-(grid / houseUsage)`.

Based on the contents of https://github.com/alpriest/EnergyStats-Android/blob/9d4b7f4874469616ca0d286c4308f38c4a0a5163/app/src/main/java/com/alpriest/energystats/ui/statsgraph/SelfSufficiencyCalculator.kt#L24-L34 I think it is just the descriptions that are incorrect and need swapping.

Perhaps it's just the descriptions need swapping in https://github.com/alpriest/EnergyStats-Android/blob/9d4b7f4874469616ca0d286c4308f38c4a0a5163/app/src/main/res/values/strings.xml#L110-L111, or perhaps it's the naming of the variables themselves - I haven't been able to quite trace it all through, otherwise I'd open a PR 😄

alpriest, 2023-08-25:
Thanks, yes just the strings needed swapping. Only affects GB and only Android too 🤷  copy/paste failure I guess.

Fixed in ee73957448c1625db8ff0ccfd870b6b98e0819cc which will be in the next release.

closed 2023-08-25 by alpriest

spco, 2023-08-25:
Thanks so much!