alpriest/EnergyStats-Android #15: PV Power Now

TonyM1958, 2023-08-26:
Hi Al,

Noticed a post on the group where you said current PV Power was not available for the power flow?

There is a variable pvPower that you use to get raw data and this returns the current PV Power being generated. Here's an example of the response:

[image]

Is this what you need to match the Fox app?

alpriest, 2023-08-26:
Thank Tony, for some reason I was using the following calculation which I think should be used when calculating historic data from the reports API rather than the raws API.

currentSolarPower = max(0, raws.currentValue(for: "batChargePower") - raws.currentValue(for: "batDischargePower") - raws.currentValue(for: "gridConsumptionPower") + raws.currentValue(for: "loadsPower") + raws.currentValue(for: "feedinPower"))

Switched in 4a29177077b059639f47389e1ef7736ea40a5206 to use pvPower and it matches Fox now. Will be in the next release.

closed 2023-08-26 by alpriest

TonyM1958, 2023-08-26:
Thanks - just checking to see this works for me. I've updated to Android v1.56 from play store but I think it might be a later version or be lagging behind the app by 5 minutes? Sometimes it seems to take a while for updates to propgate...

For info - I get the raw pvPower and integrate this over the day to get a more accurate result for the actual input pv generation than the Fox Yield figure that is the output from the inverter. This is the data I upload to PV Output.

alpriest, 2023-08-26:
Will be v1.57 next week.