alpriest/EnergyStats-Android #17: Rate limiting for Fox ESS Cloud

TonyM1958, 2023-09-15:
Don't know if this is new, but I've started to get errno 41809 when accessing the Fox cloud.

It looks like they may have implemented rate limiting - not sure if this is related to logins or calls.

Just something to watch out for in case you also starting seeing this more?

alpriest, 2023-09-15:
Thanks.

The app already copes if it gets the 41809 rate limit (shows the error message). I've also recently introduced a caching layer which I can tweak if we start to see it.

Hopefully they use a leaky bucket algorithm so shouldn't be an issue as the app is bursty, but not a heavy user. (Does less network calls than their own!)

closed 2023-09-15 by alpriest

TonyM1958, 2023-09-15:
The difference I think I am seeing is you get 41809 returned saying the token is invalid. You have to logout and log back in again but it appears it may block the login for a period.

There was about 30 seconds today when the web site went down today and returned a DNS error between 2pm and 3pm and I've seen this since then.

Hopfully, you won't be affected. I may be getting this because I'm testing stuff and having to reload data each time I update the code.

alpriest, 2023-09-15:
Shame they don’t return 40400 or 40401 which are intended for rate limiting! The apps will attempt a single token refresh on 41808/9/10 then fail with a retry button.Good to know they’re applying this now though in case it crops up on FB.For local development I use Proxyman to intercept the calls and return local results. Been very helpful for the odd infinite request loop during development!

TonyM1958, 2023-09-15:
Narrowed this down a bit - they appear to limiting the number of logins rather than the number of calls. If I store the token I get and re-use this after refreshing my code, it responds normally. But. if I drop the token and login in again, I get 41809. Doesn't entirely make sense as I'm asking for a new token.

So, from the behaviour I see, it looks like they might have implemented a minimum token validity period as well as a maximum?

alpriest, 2023-09-15:
Yes as you say sounds like they’ve rate limited logins. The app handles that as I only request a login when a token has failed.To be honest their login mechanism is pretty basic and open to MITM attacks quite easily with credentials be sent around so frequently. I wouldn’t be surprised for them to shift to an oauth login which would render my app (and all other integrations) unable to login.